xairy/raw-gadget #4: raw-gadget for linux-at91 kernel

halfguru, 2020-03-04:
Hi,

I'm trying to create a usb device gadget with the Microchip [linux-at91](https://github.com/linux4sam/linux-at91) kernel. I want to use it with a physicall usb controller and have replaced the `driver_name` and `device_name` to `atmel_usba_udc` and `500000.gadget` respectively. I am not sure if I have the correct `driver_name` but I found it in `/sys/class/udc/500000.gadget/device/uvent` displaying `DRIVER=atmel_usba_udc`.

I am running the [keyboard example](https://github.com/xairy/raw-gadget/blob/master/examples/keyboard.c) code which works well with the [dummy_hcd module](https://github.com/xairy/raw-gadget/tree/master/dummy_hcd. However, when I try to use it with the microchip usb controller, I get the following logging message:

```
event: connect, length: 0
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 64
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  descriptor = USB_DT_DEVICE
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  descriptor = USB_DT_DEVICE
ioctl(USB_RAW_IOCTL_EP0_WRITE): Connection reset by peer
```

which indicates a write failure with a `USB_REQ_GET_DESCRIPTOR` request with type `USB_DT_DEVICE`.

xairy, 2020-03-06:
Hm, it's not immediately obvious to me what's wrong here. Do you see any errors in `dmesg`? What kind of board do you use to test this?

xairy, 2020-05-30:
Please retest with the updated module and reopen if the issue still exists.

closed 2020-05-30 by xairy