xairy/raw-gadget #7: ioctl_ep_read does not return length of data read

philhug, 2020-05-03:
ioctl_ep_read does not update the length field.

It only copies back to user space the data buffer and not the header with the length field
```
        ret = copy_to_user((void __user *)(value + sizeof(io)), data, length);
```

Same for ioctl_ep0_read

xairy, 2020-05-04:
The fix is upstream, I've updated the module in the repo. Thanks!

(There are also some other fixes pending that will actually slightly change the interface.)

xairy, 2020-05-04:
Nevermind, the fix is not upstream yet, should be merged soon.

xairy, 2020-05-04:
For now you can find the fixes in this tree:

https://github.com/xairy/linux/tree/raw-gadget

philhug, 2020-05-04:
Thanks alot!

xairy, 2020-05-19:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=6e507644209b4a043363e165855a3a3050c4d40e

closed 2020-05-19 by xairy