xairy/raw-gadget #9: Facedancer backend

xairy, 2020-05-30:
It should be possible to use Raw Gadget as a backend for [Facedancer](https://github.com/usb-tools/Facedancer). At least for UDCs with fully configurable gadget endpoints.

trevor403, 2020-07-07:
This would be quite interesting, it would be great to have accessible tools for teaching USB, like the PI Zero or even the some Intel Atom and Quark boards with UDC capabilities.

xairy, 2022-12-09:
[kirill9617](https://github.com/kirill9617) implemented a [prototype](https://github.com/kirill9617/Facedancer/tree/rawgadget) Facedancer backend based on Raw Gadget.

Some notes:

1. Facedancer processes all requests in a single `service_irqs()` handler, which is incompatible with Raw Gadget's blocking read/write ioctls. A potential solution is to add timeouts for the ioctls.

2. Running some of Facedancer examples fails, the failures looks similar to https://github.com/xairy/raw-gadget/issues/29.

xairy, 2023-05-07:
Finished up the prototype and made most Facedancer examples work with Dummy and `dwc2` UDCs, see the details [here](https://github.com/xairy/raw-gadget/tree/master#facedancer-backend).