xairy/raw-gadget #11: Check for zero value in raw_ioctl_run

nickray, 2020-06-26:
Hi

I ran into issue https://github.com/nix-rust/nix/issues/781 when working on using this from Rust. The issue seems to be that `raw_ioctl_run` is declared as IO (i.e. "none"), but still checks that the passed value is zero: https://github.com/xairy/raw-gadget/blob/1e150b7a68ade7b68b0e635046ee8a15d2a72c9b/raw_gadget/raw_gadget.c#L501-L502

Could this check be removed?

xairy, 2020-06-26:
Ensuring that 0 is passed to an ioctl that shouldn't take any argument seems to be a sane thing to do.

Using `_IO*` when defining ioctl numbers is just a [convention](https://www.kernel.org/doc/Documentation/ioctl/ioctl-number.txt). There's a lot of (especially legacy) ioctls that [don't follow it at all](https://elixir.bootlin.com/linux/v5.7/source/include/uapi/asm-generic/ioctls.h#L20). On top of that there are `_IO()` ("none" as you call them) ioctls that actually [write to userspace](https://elixir.bootlin.com/linux/v5.7/source/fs/ioctl.c#L725) or [read from userspace](https://elixir.bootlin.com/linux/v5.7/source/fs/ioctl.c#L548).

Are those impossible to use from Rust? That seems weird.

Is there no way in Rust to explicitly provide all arguments to an ioctl?

nickray, 2020-06-26:
Honestly I think that library is wrong and you're right. Thanks for clarifying!

closed 2020-06-26 by nickray