xairy/raw-gadget #12: Clear EP FIFO

msbroadf, 2020-07-10:
Is it possible to add an ioctl to get status/clear fifo for an endpoint? Essentially passing these functions through 

int usb_ep_fifo_status(struct usb_ep *ep);
void usb_ep_fifo_flush(struct usb_ep *ep);

xairy, 2020-07-10:
It should be possible. What's the use case for this?

msbroadf, 2020-07-11:
Actually i ended up implementing it quickly myself and just testing to see if it fixed an error i was having, but it didnt resolve it

This call is in gadgetfs and not in raw-gadget so i suspect there will probably be a request sooner or later for it anyway

xairy, 2020-07-11:
Those ioctls are in GadgetFS, but I'm wondering what's the actual use case for them.

msbroadf, 2020-07-11:
Unfortunately im not exactly sure, im still learning the gadget subystem myself (i usually work on drivers on the other (host) side). 

Im using raw-gadget to help passthrough a USB from a remote machine transparently and it was jamming sometimes so i figured this call might be the missing part.