xairy/raw-gadget #13: Timeout for EVENT_FETCH

dheatovwil, 2020-09-17:
I have a thread for ep0_loop and another for ep_int_in_loop, apparently I can't write when `EVENT_FETCH` is blocking, is there a way to timeout the fetch, so I can write something in between?

xairy, 2020-09-17:
I'm not sure I understand. Writing to a non-zero endpoint should be possible while endpoint 0 is blocked on `USB_RAW_IOCTL_EVENT_FETCH`, see [tests/gadget.c](https://github.com/xairy/raw-gadget/blob/master/tests/gadget.c) as an example. What exactly are you trying to do?

dheatovwil, 2020-09-17:
Thank you for the fast reply and your reaffirmation!  I was trying to send a control transfer, just realized I got all the endpoints messed up.  After I changed it to write to ep0 for control packet it seems to be working !  Thanks once again =D

closed 2020-09-17 by dheatovwil