xairy/raw-gadget #22: raw gadget can be use to right mass storage device

Avni22, 2022-02-22:
Hi

I am exploring the raw gadget driver for masss storage with dummy_hcd driver. 

I am getting the raw data in user space for mass storage with dummy_hcd. I want to pass that data to target pc using OTG gadget driver.

is there any way to use these raw usb as bridge between them?  

and can we read/wright directly to pendrive or sd card with raw driver? 
if you have any example for these can you please point me.  any duhelp will be appreciable. 

Thanks

xairy, 2022-02-22:
Hi,

I don't understand what you want to do.

> I am getting the raw data in user space for mass storage with dummy_hcd. I want to pass that data to target pc using OTG gadget driver.

So you have a program that emulates a USB mass-storage device and works with dummy_hcd, right? Then you likely only need to use a hardware UDC instead of dummy_hcd.

> is there any way to use these raw usb as bridge between them?

Don't understand this part. A bridge between what?

> and can we read/wright directly to pendrive or sd card with raw driver?

No. Raw Gadget is for emulating USB devices. You can have an SD card connected to the hardware emulating a USB device via Raw Gadget, but Raw Gadget has nothing to do with it.

xairy, 2022-03-08:
@AristoChen Awesome project! Happy to see Raw Gadget in use! :)

AristoChen, 2022-03-08:
@xairy I noticed you have shared it on your twitter, Thanks!

xairy, 2022-05-21:
This seems to be resolved. Please reopen and update if not.

closed 2022-05-21 by xairy