xairy/raw-gadget #24: host to device transfer size

lmirel, 2022-04-08:
Hi, I've tested on Raspberry PI 4 and BeagleBone Black (using the [usb-proxy](https://github.com/AristoChen/usb-proxy/issues/1) project) and on both there seems to be an issue with data from host to device.
It seems that when we get data from the host, it could contain more than just one packet (up to nfo.eps[i].limits.maxpacket_limit) which aside from the weird behavior on the actual device, it may also get lost when sent back to the device.

Any chance this can be tuned to only process one message at a time?

On Raspberry PI 4 the behavior is worse as it transfers 1024 bytes at a time most of the time and sometimes it gets blocked for a second with no transfer at all but the Raspberry PI is somewhat known to exhibit USB issues.

The transfers from device to host don't show any issue and they respect the expected message size.

Thanks in advance for any feedback

xairy, 2022-05-21:
Hi,

Hm, I believe this is functioning as intended.

One `USB_RAW_IOCTL_EP_READ` call queues one request via `usb_ep_queue()`. However, the request might be filled with multiple packets. See the [comment](https://elixir.bootlin.com/linux/v5.18-rc7/source/drivers/usb/gadget/udc/core.c#L223) for `usb_ep_queue()` and [its implementation](https://elixir.bootlin.com/linux/v5.18-rc7/source/drivers/usb/gadget/udc/net2280.c#L705) for net2280 (as an example).

I don't think we need any changes in Raw Gadget. However, USB proxy needs to be aware of this behavior.

Thanks!

closed 2022-05-21 by xairy

lmirel, 2022-05-21:
Thanks, I adapted the USB proxy implementation to process one packet at a time based on endpoint wMaxPacketSize and it works as expected. The flow fails on Rapsberry PI4 after a while and I was hoping there was something that could be done on the kernel driver side but couldn't find anything useful so far.