xairy/raw-gadget #25: Raw-gadget ioctls may access uninitialized data

ryan-summers, 2022-05-05:
For the IOCTLs of the `raw-gadget` module that do not accept any data (e.g. `raw_ioctl_run`), the `value` argument of the IOCTL call is uninitialized (variadic arg to `ioctl()`) and should not be read.

This causes issue when using cross-language bindings to IOCTL into the `raw-gadget` driver, such as using `nix` from Rust, which intentionally omits any variadic arguments to ioctl since the call requires no data. This causes these functions to spuriously EINVAL as an error code.

ryan-summers, 2022-05-05:
For context: I'm looking to leverage `raw-gadget` as a break-in to the linux USB stack and want to use it for entire-host side testing of the [Rust embedded `usb-device`](https://github.com/rust-embedded-community/usb-device) peripheral-side USB stack.

xairy, 2022-05-08:
Hi,

This appears to be the same issue as in #11. The `_IO*` definition of an ioctl number do not formally specify the types of arguments this ioctl accepts. `_IO()` ioctls can and [do](https://github.com/xairy/raw-gadget/issues/11#issuecomment-650437417) take arguments. I believe the documentation for this nix library [offers](https://docs.rs/nix/latest/nix/sys/ioctl/index.html#using-bad-ioctls) a solution for this.

Thanks!

closed 2022-05-08 by xairy