xairy/raw-gadget #26: Fail to compile raw-gadget after execute update.sh

AristoChen, 2022-07-24:
Hi,

the following error message will appear if trying to compile `raw-gadget` kernel module after `update.sh` is executed
```shell
$ make
make -C /lib/modules/5.10.60/build M=/root/raw-gadget/raw_gadget SUBDIRS=/root/raw-gadget/raw_gadget modules
make[1]: Entering directory '/usr/src/linux-source'
  CC [M]  /root/raw-gadget/raw_gadget/raw_gadget.o
/root/raw-gadget/raw_gadget/raw_gadget.c: In function ‘raw_ioctl_run’:
/root/raw-gadget/raw_gadget/raw_gadget.c:546:8: error: implicit declaration of function ‘usb_gadget_register_driver’; did you mean ‘usb_gadget_unregister_driver’? [-Werror=implicit-function-declaration]
  546 |  ret = usb_gadget_register_driver(&dev->driver);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |        usb_gadget_unregister_driver
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /root/raw-gadget/raw_gadget/raw_gadget.o] Error 1
make[1]: *** [Makefile:1824: /root/raw-gadget/raw_gadget] Error 2
make[1]: Leaving directory '/usr/src/linux-source'
make: *** [Makefile:7: default] Error 2
```

looks like this [patch](https://lore.kernel.org/all/YmSc29YZvxgT5fEJ@rowland.harvard.edu/) is causing the issue, `usb_gadget_probe_driver()` was renamed to `usb_gadget_register_driver()`,  the following patch works fine with my SBC with `5.10.60` kernel version.

```patch
diff --git a/raw_gadget/update.sh b/raw_gadget/update.sh
index 6ffcb66..c363e12 100755
--- a/raw_gadget/update.sh
+++ b/raw_gadget/update.sh
@@ -9,4 +9,12 @@ BRANCH=usb-next
 wget $REPO/drivers/usb/gadget/legacy/raw_gadget.c?h=$BRANCH -O raw_gadget.c
 wget $REPO/include/uapi/linux/usb/raw_gadget.h?h=$BRANCH -O raw_gadget.h

+
+HEADER_FILE=/lib/modules/"$(uname -r)"/build/include/linux/usb/gadget.h
+if [[ $(grep -c "usb_gadget_probe_driver(" < "$HEADER_FILE") -ne 0 ]]
+then
+    echo "Replacing from usb_gadget_register_driver() to usb_gadget_probe_driver()"
+    sed -i 's/usb_gadget_register_driver/usb_gadget_probe_driver/g' raw_gadget.c
+fi
+
 git apply ./include.patch

```

xairy, 2022-07-27:
Hi,

Yeah, this is expected. There is no stable API for kernel modules. I think, for simplicity, we should keep the code in this repository compatible with the latest R Pi Zero kernel. And add a note what to do in case the modules do not compile.

I'll take a closer look into doing this when I get a chance. Let's keep this bug open for now.

Thanks!

xairy, 2022-12-08:
Fixed by https://github.com/xairy/raw-gadget/commit/8274ee02badc8804e63cca41ed24fbc223ff69f9.

closed 2022-12-08 by xairy

pr8x, 2023-08-01:
I am on kernel version 6.1.21 (Raspbian/RPi Zero) and I am seeing the same issue. `usb_gadget_probe_driver` doesn't seem to exist. I pulled the kernel headers via `raspberrypi-kernel-headers`. Is there anything I am missing here?

xairy, 2023-08-01:
You need to comment out [this part](https://github.com/xairy/raw-gadget/blob/master/raw_gadget/update.sh#L16) and reupdate the module.

pr8x, 2023-08-01:
Ah, got it. Thanks.