xairy/raw-gadget #27: `make: *** [Makefile:7: default] Error 2`

Marvin-Brouwer, 2022-07-27:
Hi,

I'm having the hardest time to get my Raspi Zero to show up as a USB.
I found a couple of blogs pointing towards this github so I figured I'd try to install `dummy_hcd` since it is in fact missing in my system.
I had some cpp issues with fallthrough, so now I've updated to the latest version of CPP.

If I now run make like this to make sure it uses the correct cpp version:
```bash
cpp_dir="/opt/gcc-10.1.0/bin";
sudo make CC="${cpp_dir}/gcc-10.1" CPP="${cpp_dir}/g++-10.1" CXX="${cpp_dir}/g++-10.1" LD="${cpp_dir}/g++-10.1";
```
I still get the following error:
```text
make -C /lib/modules/4.19.97+/build M=/usr/src/linux-headers-4.19.97+/dummy_hcd SUBDIRS=/usr/src/linux-headers-4.19.97+/dummy_hcd modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.97+'
  CC [M]  /usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.o
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c: In function ‘dummy_enable’:
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c:556:4: error: ‘fallthrough’ undeclared (first use in this function)
  556 |    fallthrough;
      |    ^~~~~~~~~~~
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c:556:4: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c: In function ‘dummy_timer’:
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c:1971:4: error: ‘fallthrough’ undeclared (first use in this function)
 1971 |    fallthrough;
      |    ^~~~~~~~~~~
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c: In function ‘dummy_hub_control’:
/usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.c:2148:4: error: ‘fallthrough’ undeclared (first use in this function)
 2148 |    fallthrough;
      |    ^~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:310: /usr/src/linux-headers-4.19.97+/dummy_hcd/dummy_hcd.o] Error 1
make[1]: *** [Makefile:1522: _module_/usr/src/linux-headers-4.19.97+/dummy_hcd] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.97+'
make: *** [Makefile:7: default] Error 2
```

Do you have any suggestions on what I can try to debug this some more?

xairy, 2022-07-27:
Hi,

I think the RPi kernel is just old and does not define the fallthrough pesudo-keyword.

You can try adding the definition into `dummy_hcd.c`:

``` c
#if __has_attribute(__fallthrough__)
# define fallthrough                    __attribute__((__fallthrough__))
#else
# define fallthrough                    do {} while (0)  /* fallthrough */
#endif
```

Thanks!

closed 2022-07-27 by xairy

Marvin-Brouwer, 2022-07-28:
The make file is giving me this now:
```txt
 In function ‘dummy_enable’:
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:618:6: warning: missing terminating " character
  618 |      "non-bulk ep %s
      |      ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:619:1: warning: missing terminating " character
  619 | ", _ep->name);
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:626:24: warning: missing terminating " character
  626 |  dev_dbg(udc_dev(dum), "enabled %s (ep%d%s-%s) maxpacket %d stream %s
      |                        ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:627:1: warning: missing terminating " character
  627 | ",
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:674:24: warning: missing terminating " character
  674 |  dev_dbg(udc_dev(dum), "disabled %s
      |                        ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:675:1: warning: missing terminating " character
  675 | ", _ep->name);
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:735:24: warning: missing terminating " character
  735 |  dev_dbg(udc_dev(dum), "ep %p queue req %p to %s, len %d buf %p
      |                        ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:736:1: warning: missing terminating " character
  736 | ",
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:801:5: warning: missing terminating " character
  801 |     "dequeued req %p from %s, len %d buf %p
      |     ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:802:1: warning: missing terminating " character
  802 | ",
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:985:35: warning: missing terminating " character
  985 |  return scnprintf(buf, PAGE_SIZE, "%s
      |                                   ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:986:1: warning: missing terminating " character
  986 | ", dum->driver->function);
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1020:31: warning: missing terminating " character
 1020 |   dev_err(dummy_dev(dum_hcd), "Unsupported driver max speed %d
      |                               ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1021:1: warning: missing terminating " character
 1021 | ",
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1150:22: warning: missing terminating " character
 1150 |  dev_dbg(&pdev->dev, "%s
      |                      ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1151:1: warning: missing terminating " character
 1151 | ", __func__);
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1162:22: warning: missing terminating " character
 1162 |  dev_dbg(&pdev->dev, "%s
      |                      ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1163:1: warning: missing terminating " character
 1163 | ", __func__);
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1270:31: warning: missing terminating " character
 1270 |   dev_err(dummy_dev(dum_hcd), "Stream id %d is out of range.
      |                               ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1271:1: warning: missing terminating " character
 1271 | ",
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1314:31: warning: missing terminating " character
 1314 |   dev_err(dummy_dev(dum_hcd), "usb_device address has changed!
      |                               ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1315:1: warning: missing terminating " character
 1315 | ");
      | ^
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:1651:25: warning: missing terminating " character
[... 145 lines cut ...]
      |           ^
      |           ;
......
 2958 | module_exit(cleanup);
      |
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:617:4: error: expected declaration or statement at end of input
  617 |    dev_err(udc_dev(dum), "Can't enable stream support on "
      |    ^~~~~~~
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:617:4: error: expected declaration or statement at end of input
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:617:4: error: expected declaration or statement at end of input
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:611:3: error: label ‘done’ used but not defined
  611 |   goto done;
      |   ^~~~
At top level:
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:504:12: warning: ‘dummy_enable’ defined but not used [-Wunused-function]
  504 | static int dummy_enable(struct usb_ep *_ep,
      |            ^~~~~~~~~~~~
/home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.c:434:13: warning: ‘set_link_state’ defined but not used [-Wunused-function]
  434 | static void set_link_state(struct dummy_hcd *dum_hcd)
      |             ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:310: /home/●●●●●/install/git/raw-gadget/dummy_hcd/dummy_hcd.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/●●●●●/install/git/raw-gadget/dummy_hcd] Error 2
```
I'm not a C++ programmer myself so I'm not really sure where to start with this.
Could you maybe give me a pointer in the right direction to where I can figure this out myself?

xairy, 2022-07-29:
Based on the error about `dev_err` not being defined, I think there is something wrong with your kernel headers. Otherwise, no idea.

Marvin-Brouwer, 2022-07-29:
Right, that sounds like a lot of effort.
Honestly, I don't necessarily need Raspbian.
Do you maybe have another debian based distro to recomment?
At the end of the day I just want my raspi to emulate a USB or a preferably a CDROM when I plug it in.

xairy, 2022-07-29:
I don't have any recommendation wrt a Debian distro for R Pi Zero.

But, on a related note, I never had any issues compiling the Dummy HCD module on R Pi Zero. Clean install + cloning this repo + building the module with the provided scripts always worked for me.

Marvin-Brouwer, 2022-07-29:
Hmm that's interesting.
All I did was:

- Mount the image
- Chroot some stuff on the image
- Clone this repo into the image.
- Flash onto the pi.
- Connect via COM and call make