xairy/raw-gadget #29: how to send configuration descriptor in 2 parts

Alphadan28, 2022-11-03:
first of all thanks for sharing your amazing work.

I'm trying to send a configuration descriptor which has a size of 59 bytes to a device which has bMaxPacketSize0 of 32 bytes.

if i try to send the full 59 bytes i get this

Sending data to EP0(control_in): 09 02 3b 00 02 01 00 a0 32 09 04 00 00 01 03 01 01 00 09 21 01 01 00 01 22 4d 00 07 05 81 03 20 00 08 09 04 01 00 01 03 01 02 00 09 21 01 01 00 01 22 74 00 07 05 82 03 20 00 01
ioctl(USB_RAW_IOCTL_EP0_WRITE): Cannot send after transport endpoint shutdown

whats the right way to send this descriptor to that device is this possible using raw gadget or where can i get info about this.

xairy, 2022-11-04:
Hi @Alphadan28, thank you!

I'm not sure what the issue could be.

You mean that the device controller driver that you use only allows `bMaxPacketSize0` up to 32 bytes, right? Which device controller is this?

xairy, 2022-11-04:
You're using a Raspberry Pi, right? Looking at the dwc2 driver source, it should return `-ESHUTDOWN` when the device is being disconnect. I'm not sure why this happens though.

Let's check if this is the case: could you add `dev_err` log messages into `gadget_disconnect` and the neighboring functions and see if any of those are visible in `dmesg`?

xairy, 2022-11-04:
FTR: The issue originated from https://github.com/AristoChen/usb-proxy/issues/4.

Alphadan28, 2022-11-04:
yes is a raspberry pi4, sure i can make those tests but excuse my ignorance in this maters but you mean something like this right?

/* These are currently unused but present in case UDC driver requires them. */
`static void gadget_disconnect(struct usb_gadget *gadget) {
				dev_err(&gadget->dev, "Gadget Disconect\n");
}`
`static void gadget_suspend(struct usb_gadget *gadget) { 
	            dev_err(&gadget->dev, "Gadget Suspend\n");
}`
`static void gadget_resume(struct usb_gadget *gadget) {
	           dev_err(&gadget->dev, "Gadget Resume\n");
}`
`static void gadget_reset(struct usb_gadget *gadget) { 
               dev_err(&gadget->dev, "Gadget Reset\n");
}`
/*----------------------------------------------------------------------*/

but I'm not sure where to read those inputs

Alphadan28, 2022-11-04:
> FTR: The issue originated from [AristoChen/usb-proxy#4](https://github.com/AristoChen/usb-proxy/issues/4).

yes is the same issue, i made some hypothesis in the way but all got discarded by some tests i made

xairy, 2022-11-04:
> you mean something like this right?

Yes.

> but I'm not sure where to read those inputs

Run `dmesg` on the Raspberry Pi.

Alphadan28, 2022-11-04:
this is the result:

pi@raspberrypi:~/usb-proxy $ sudo ./usb-proxy --device=fe980000.usb --driver=fe980000.usb --vendor_id=1997 --product_id=2433
Device is: fe980000.usb
Driver is: fe980000.usb
vendor_id is: 6551
product_id is: 9267
Device opened successfully
Setup USB config successfully
Start hotplug_monitor thread, thread id(2264)
Start for EP0, thread id(2262)
event: connect, length: 0
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 64
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
ioctl(USB_RAW_IOCTL_EP0_WRITE): Cannot send after transport endpoint shutdown

and the dmesg output:
[ 3080.423910] raw-gadget.0 gadget: Gadget Disconect
[ 3080.429020] dwc2 fe980000.usb: new device is full-speed
[ 3080.429051] raw-gadget.0 gadget: Gadget Disconect

indeed gadget is getting disconected, not sure what that means though.

xairy, 2022-11-04:
I see. Could you also try what I suggested in https://github.com/AristoChen/usb-proxy/issues/4#issuecomment-1302876212?

Alphadan28, 2022-11-04:
> I see. Could you also try what I suggested in [AristoChen/usb-proxy#4 (comment)](https://github.com/AristoChen/usb-proxy/issues/4#issuecomment-1302876212)?

sure this is the result: connecting as USB_SPEED_FULL instead of USB_SPEED_HIGH

Device is: fe980000.usb
Driver is: fe980000.usb
vendor_id is: 6551
product_id is: 9267
Device opened successfully
Start hotplug_monitor thread, thread id(2631)
Setup USB config successfully
Start for EP0, thread id(2629)
event: connect, length: 0
event: control, length: 8
bRequestType: 0x80 (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 64
type = USB_TYPE_STANDARD
req = USB_REQ_GET_DESCRIPTOR
desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
bRequestType: 0x80 (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
type = USB_TYPE_STANDARD
req = USB_REQ_GET_DESCRIPTOR
desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
bRequestType: 0x80 (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
type = USB_TYPE_STANDARD
req = USB_REQ_GET_DESCRIPTOR
desc = USB_DT_CONFIG
ioctl(USB_RAW_IOCTL_EP0_WRITE): Cannot send after transport endpoint shutdown

and dmesg output:

[ 3458.920575] hwmon hwmon1: Undervoltage detected!
[ 3467.240672] hwmon hwmon1: Voltage normalised
[ 3471.400790] hwmon hwmon1: Undervoltage detected!
[ 3483.880773] hwmon hwmon1: Voltage normalised
[ 3603.254003] dwc2 fe980000.usb: bound driver raw-gadget.0
[ 3603.433015] dwc2 fe980000.usb: new device is full-speed
[ 3603.487987] dwc2 fe980000.usb: new device is full-speed
[ 3603.522760] dwc2 fe980000.usb: new address 45
[ 3604.042229] raw-gadget.0 gadget: Gadget Disconect
[ 3604.047030] dwc2 fe980000.usb: new device is full-speed
[ 3604.047062] raw-gadget.0 gadget: Gadget Disconect

xairy, 2022-11-04:
So doesn't help... My guess was that since only Full Speed devices can have MaxPacketSize0 of 32 bytes, the UDC might reject this size for a High Speed device. But if this is not the case, I don't know what the issue could be. One way to debug this is to understand what exactly GadgetFS and the GadgetFS-based proxy do differently. But there's no simple test to figure it out.

Alphadan28, 2022-11-04:
yes very sneaky problem, but with my mouse manages to send the descriptor which is bigger than the maxpacketsize0 but it crashes when trying to send the interrupt with mouse data as soon as i move the mouse the error produces so is very weird situation.

AristoChen, 2022-11-04:
Hi @Alphadan28, 

I noticed that you have tried using another [USBProxy](https://github.com/nesto-software/USBProxy) project to test. Actually my usb-proxy was inspired by its upstream [repo](https://github.com/usb-tools/USBProxy-legacy), so I think I might be able to answer some of your questions, though I might not be correct lol

> just my perception is that raw gadget is lot faster than GadgetFS

It is actually possible to speed up the GadgetFS based USBProxy project, you can try to combine several USB packets and send them together. AFAIK, the hardware itself will split those data into small pieces which can fit in the maxPacketSize, so you don't need to worry about the hardware will crash, and it should be able to handle the split a lot faster than software

> what I found in the pcap was that the mini keyboard shows in the device descriptor MaxPacketSize0 = 64 which is weird because that device conected directly reports its MaxPacketSize0 = 32 and also usb.bcdDevice = 0x0200 instead of usb.bcdDevice = 0x0110 (which is its original value) its like that other proxy somehow changes these values.

This was from your comment in my project issue. IIRC, the kernel module did this, but sadly I forget which function implement this. According to my previous experience, this part does not affect all the USB devices(at least dozens of USB devices) that I have tested with the GadgetFS based USBProxy

> USB headset (it sends all descriptors it doesn't crash but doesn't transfer any audio but the volume slider generates traffic that goes through the proxy)

I guess the USB headset should be using isochronous transfer for the audio, it is not supported in raw-gadget, so my project won't be able to support. IIRC, it should be not supported in GadgetFS based USBProxy as well. If one day the raw-gadget does support the isochronous transfer, then I will find some time to implement it in my project

Alphadan28, 2022-11-04:
> I guess the USB headset should be using isochronous transfer for the audio, it is not supported in raw-gadget, so my project won't be able to support. IIRC, it should be not supported in GadgetFS [...]
**ohh i did not know this**

>the hardware itself will split those data into small pieces which can fit in the maxPacketSize, so you don't need to worry about the hardware will crash, and it should be able to handle the split a lot faster than software.

about this i did not know it was being made in hardware level but that clears lot of doubs i had since reading usb specs i saw the packets have a structure as described here: https://www.beyondlogic.org/usbnutshell/usb3.shtml

and that was surely being handled in lower levels because wasn't able to find anything about these fields, I noticed that when the mouse sent the descriptor which was bigger than maxpacketsize0, i wish there was more info around about this topics is there something i can read to know more about raw gadget i did not know for example it did not support isochronous endpoints is it there any documentation about this or just by looking at the code is the only way?

AristoChen, 2022-11-07:
>  is it there any documentation about this

This [document](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/usb/raw-gadget.rst#n93) might be the thing that you want

xairy, 2023-05-06:
I did some testing and it looks like this is now a Raw Gadget bug.

For those affected: please retest with https://github.com/AristoChen/usb-proxy/pull/6.

closed 2023-05-06 by xairy