xairy/raw-gadget #31: missing -EBUSY inside raw_process_ep_io and raw_process_ep0_io

patryk4815, 2022-12-05:
Hi.
After implementing my proxy( https://github.com/patryk4815/usb-proxy ) I tried to debug why "sometime" my code fails with proxying mass storage.
It was really "random", so after hours of debugging my code, it looks like the error is your module :(

ftrace from kernel:
```code
       usb_proxy-5634    [000] ..... 35373.583990: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [001] ..... 35373.584716: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [002] ..... 35373.585358: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [000] ..... 35373.585993: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [000] ..... 35373.586636: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [002] ..... 35373.587320: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [002] ..... 35373.587994: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.588683: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.589416: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.590023: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.590693: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.591659: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.591701: usb_ep_dequeue+0x4/0xa8 <-raw_process_ep_io+0x198/0x4d0 [raw_gadget]
       usb_proxy-5634    [003] ..... 35373.591740: usb_ep_queue+0x4/0xc8 <-raw_process_ep_io+0x164/0x4d0 [raw_gadget]
```
After some time it randomly get `usb_ep_dequeue`.
So i looked into `raw-gadget` code.

https://github.com/xairy/raw-gadget/blob/master/raw_gadget/raw_gadget.c#L1044
```
	ret = usb_ep_queue(ep->ep, ep->req, GFP_KERNEL);
	if (ret) {
		dev_err(&dev->gadget->dev,
				"fail, usb_ep_queue returned %d\n", ret);
		spin_lock_irqsave(&dev->lock, flags);
		dev->state = STATE_DEV_FAILED;
	        // MISSING "ret = -EBUSY"?
		goto out_done;
	}

	ret = wait_for_completion_interruptible(&done);
	if (ret) {
		dev_dbg(&dev->gadget->dev, "wait interrupted\n");
		usb_ep_dequeue(ep->ep, ep->req);
		wait_for_completion(&done);
		spin_lock_irqsave(&dev->lock, flags);
                 // MISSING "ret = -EBUSY"?
		goto out_done;
	}
```
I think there is missing `ret = -EBUSY`. Not sure. 
But from `USB_RAW_IOCTL_EP_WRITE` or `USB_RAW_IOCTL_EP_READ` i should get response about "error" right? 
Now im getting "good" length writted/readed.

patryk4815, 2022-12-05:
Hmm, probably may issue is not abount missing `ret` 🤔 . Now I see `wait_for_completion_interruptible` should return error 🤔

patryk4815, 2022-12-05:
I wrote this patch and now everything working fine:
[image]

patryk4815, 2022-12-06:
btw i asked GPT where error 👀 
[image]

xairy, 2022-12-08:
Hi,

There's indeed an issue with returning a proper code when the request was interrupted, we probably need a fix for this.

However, I don't understand why the issue you mention is happening. `usb_ep_dequeue` is not checked, as it can fail in case the request has already been dequeued. And then its completion routine must have already been executed.

However, let's check if `usb_ep_dequeue` indeed fails.

Could you try running your test with the following patch:

``` c
diff --git a/raw_gadget/raw_gadget.c b/raw_gadget/raw_gadget.c
index f203fe2..a294589 100644
--- a/raw_gadget/raw_gadget.c
+++ b/raw_gadget/raw_gadget.c
@@ -968,6 +968,8 @@ static void gadget_ep_complete(struct usb_ep *ep, struct usb_request *req)
        spin_unlock_irqrestore(&dev->lock, flags);
 
        complete((struct completion *)req->context);
+
+       dev_dbg(&dev->gadget->dev, "ep = %px, completed\n", r_ep);
 }
 
 static int raw_process_ep_io(struct raw_dev *dev, struct usb_raw_ep_io *io,
@@ -1035,10 +1037,14 @@ static int raw_process_ep_io(struct raw_dev *dev, struct usb_raw_ep_io *io,
                goto out_done;
        }
 
+       dev_dbg(&dev->gadget->dev, "ep = %px, queued\n", ep);
+
        ret = wait_for_completion_interruptible(&done);
        if (ret) {
                dev_dbg(&dev->gadget->dev, "wait interrupted\n");
-               usb_ep_dequeue(ep->ep, ep->req);
+               int deq_ret = usb_ep_dequeue(ep->ep, ep->req);
+               if (deq_ret)
+                       dev_dbg(&dev->gadget->dev, "ep = %px, deq_ret = %d\n", ep, deq_ret);
                wait_for_completion(&done);
                spin_lock_irqsave(&dev->lock, flags);
                goto out_done;
```

and share the `dmesg` output?

Thanks!

patryk4815, 2022-12-11:
```
[...]
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: wait interrupted
Dec 11 23:54:30 rpi.cypis.ovh kernel: dwc2 fe980000.usb: dwc2_hsotg_ep_stop_xfr: timeout DIEPINT.NAKEFF
Dec 11 23:54:30 rpi.cypis.ovh kernel: dwc2 fe980000.usb: dwc2_hsotg_ep_stop_xfr: timeout DOEPCTL.EPDisable
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, completed
Dec 11 23:54:30 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849d59e71b0, queued
```

patryk4815, 2022-12-12:
Maybe can we remove `usb_ep_dequeue`? 
From kernel documentation:
```
 * Note that some hardware can't clear out write fifos (to unlink the request
 * at the head of the queue) except as part of disconnecting from usb. Such
 * restrictions prevent drivers from supporting configuration changes,
 * even to configuration zero (a "chapter 9" requirement).
 ```

patryk4815, 2022-12-12:
Sometime getting this:
```
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2, queued
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2 completed
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2, queued
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: wait interrupted ep_id=2
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2 completed
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2, deq_ret = -22
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: wait interrupted done ep_id=2
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2, queued
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2 completed
Dec 12 00:19:13 rpi.cypis.ovh kernel: raw-gadget.0 gadget: ep = ffff1849c1e76880, ep_id=2, queued
```

patryk4815, 2022-12-12:
Btw `-ERESTARTSYS` is not propagating to user-space don't know why 🦄

xairy, 2023-10-26:
Sent the fix [upstream}https://lore.kernel.org/linux-usb/0db45b1d7cc466e3d4d1ab353f61d63c977fbbc5.1698329862.git.andreyknvl@gmail.com/T/#t). Will update the `master` branch once the patch is merged.

xairy, 2023-10-26:
Note that the device emulation code needs to watch out for the `-EINTR` return code and then resend the request.

closed 2023-10-26 by xairy