xairy/raw-gadget #43: I have a card reader, use AristoChen/usb-proxy  proxy it windows pc failed

imatespl, 2023-07-22:
## 1. usb card reader info

```
root@orangepipc:~/usb-proxy# lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 004: ID 2ce3:9563 Generic EMV Smartcard Reader
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@orangepipc:~/usb-proxy# lsusb -v -d 2ce3:9563

Bus 006 Device 004: ID 2ce3:9563 Generic EMV Smartcard Reader
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x2ce3
  idProduct          0x9563
  bcdDevice            1.20
  iManufacturer           1 Generic
  iProduct                2 EMV Smartcard Reader
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005d
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.10  (Warning: Only accurate for version 1.0)
        nMaxSlotIndex           0
        bVoltageSupport         7  5.0V 3.0V 1.8V
        dwProtocols             3  T=0 T=1
        dwDefaultClock       4000
        dwMaxiumumClock     12000
        bNumClockSupported      3
        dwDataRate          10753 bps
        dwMaxDataRate      688172 bps
        bNumDataRatesSupp.     95
        dwMaxIFSD             254
        dwSyncProtocols  00000007  2-wire 3-wire I2C
        dwMechanical     00000000
        dwFeatures       000404BE
[... 209 lines cut ...]
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
Control transfer succeed
Sending data to EP0(control_in): 09 02 5d 00 01 01 00 a0 19 09 04 00 00 03 0b 00 00 00 36 21 10 01 00 07 03 00 00 00 a0 0f 00 00 e0 2e 00 00 03 01 2a 00 00 2c 80 0a 00 5f fe 00 00 00 07 00 00 00 00 00 00 00 be 04 04 00 10 01 00 00 ff ff 00 00 00 01 07 05 81 03 04 00 01 07 05 02 02 10 00 00 07 05 83 02 10 00 00

```
##4. This two file is pcap capture from orangepc and windows
[orangepc pcap](https://www.dropbox.com/scl/fi/6igpr4enf6wkdg37loqqp/card_reader_in_orangepc.pcap?rlkey=04t6zzmc1dmmqcpqb3otbg9mi&dl=0)
[windows  pcap](https://www.dropbox.com/scl/fi/f9x9j75ok3b1ihuz4lh7n/card_reader_in_windows.pcapng?rlkey=yb0yrnzd99shxmyzaxduwdilb&dl=0)
## 5. It may not raw_gadget issues, but i can not find  how to debug it, if you could help me, I would be very grateful. Thank you.

xairy, 2023-07-22:
Hi!

Could you clarify, what is the problem? USB device is not connecting to host when proxied? I don't see any errors in the log.

Could you also try proxying the device on a Raspberry Pi? I suspect something works differently on Orange boards. I don't have one myself to test.

Also,  which card reader do you use? Please give an Amazon link or something like that.

Thanks!

imatespl, 2023-07-23:
ok，I will try on  raspberry pi，the problem is loop in get device
```
  **bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 64**
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
Control transfer succeed
Sending data to EP0(control_in): 09 02 5d 00 01 01 00 a0 19 09 04 00 00 03 0b 00 00 00 36 21 10 01 00 07 03 00 00 00 a0 0f 00 00 e0 2e 00 00 03 01 2a 00 00 2c 80 0a 00 5f fe 00 00 00 07 00 00 00 00 00 00 00 be 04 04 00 10 01 00 00 ff ff 00 00 00 01 07 05 81 03 04 00 01 07 05 02 02 10 00 00 07 05 83 02 10 00 00
ep0: transferred 93 bytes (in)
event: control, length: 8
  **bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 64**
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 01 02 00 00 00 40 e3 2c 63 95 20 01 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
Control transfer succeed
Sending data to EP0(control_in): 09 02 5d 00 01 01 00 a0 19 09 04 00 00 03 0b 00 00 00 36 21 10 01 00 07 03 00 00 00 a0 0f 00 00 e0 2e 00 00 03 01 2a 00 00 2c 80 0a 00 5f fe 00 00 00 07 00 00 00 00 00 00 00 be 04 04 00 10 01 00 00 ff ff 00 00 00 01 07 05 81 03 04 00 01 07 05 02 02 10 00 00 07 05 83 02 10 00 00
```
I alse have a card reader can proxy，the console echo like this，the first get device  get config like wirte to musb-hdrc driver, and the response is musb-hdrc generic, then get device again get config again，this is send to pc by musb-hdrc，and response to raw-gadget
```
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 64
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 00 02 00 00 00 40 6b 07 21 30 02 03 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0100, wIndex: 0x0000, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
Control transfer succeed
Sending data to EP0(control_in): 12 01 00 02 00 00 00 40 6b 07 21 30 02 03 01 02 00 01
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0200, wIndex: 0x0000, wLength: 255
  type = USB_TYPE_STANDARD
[... 49 lines cut ...]
Activating 3 endpoints on interface 0
int_in: addr = 3, ep = #4
**Creating thread for EP83**
Start reading thread for EP83, thread id(28356)
bulk_in: addr = 4, ep = #6
Creating thread for EP84
Start reading thread for EP84, thread id(28358)
bulk_out: addr = 5, ep = #7
**Creating thread for EP05**
Start writing thread for EP83, thread id(28357)
Start writing thread for EP84, thread id(28359)
Start reading thread for EP05, thread id(28360)
process_eps done
event: control, length: 8
  bRequestType: 0x80  (IN), bRequest: 0x06, wValue: 0x0301, wIndex: 0x0409, wLength: 4
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
Start writing thread for EP05, thread id(28361)
EP83(int_in): enqueued 2 bytes to queue
Sending data to EP83(int_in): 50 02
```
I will confirm the amazon link

Thank you very much.

closed 2023-07-23 by imatespl

imatespl, 2023-07-23:
I understand the issue now. The problem lies with the Linux card reader driver, which may cause issues when it doesn't recognize the device driver. To avoid this problem, you can set the device identification as a known device manufacturer during the response of the device information.
```
        struct usb_device_descriptor* pdata = (struct usb_device_descriptor*)&io.data;
        pdata->bcdUSB = 0x200;
        pdata->idVendor = 0x076b;
        pdata->idProduct = 0x3021;
        pdata->bcdDevice = 0x0302;
```