xairy/raw-gadget #46: Document or change behavior with 0-length setup requests

xairy, 2023-08-19:
Currently, Raw Gadget [considers](https://elixir.bootlin.com/linux/v6.5-rc6/source/drivers/usb/gadget/legacy/raw_gadget.c#L352) all requests with `wLength == 0` to be `OUT` requests regardless of `bRequestType`.

This follows the HCD drivers' [behavior](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fea3409112a93581db18da4c4332c8bf8d68af6b) (also [this](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=6dabeb891c001c592645df2f477fed9f5d959987)), but I'm not sure if this makes sense for the gadget side.

We need to either document this behavior or change it.

dukezuoHZ, 2024-01-02:
I find this issue too,if usb_ctrlrequest carried option data, We cann't get it. how to fix this issue?

xairy, 2024-01-06:
This issue is only related to 0-length requests, which carry no data. If your request transfers data, you're having a different issue.