xairy/raw-gadget #48: Fix dwc2 issuing disconnect instead of reset

xairy, 2023-08-20:
Not a bug in Raw Gadget but still related to its usage.

dwc2 calls the gadget driver's `->disconnect()` callback when the port to which the device is connected gets reset by
the host (via `SET_FEATURE`/`PORT_RESET`). It should call `->reset()` instead.

Asked kernel maintainers about getting this fixed, see [here](https://lore.kernel.org/linux-usb/CA+fCnZfePUVqSJWBsPsBghiDQ165FhkaC38aR9GSrnY0BML5Sw@mail.gmail.com/).

anomaly256, 2024-03-18:
Hi, was there ever any progress on this?  I suspect I might be hitting it in my project.  I see an alleged device reset then disconnect.  Admittedly I don't know enough about the stack to know for sure what the issue is though

```
Mar 18 00:44:29 host usb-proxy[1435]: Resent Bulk transfer on EP02 for attempt 1. length(1024), transferred(1024)
Mar 18 00:44:29 host usb-proxy[1435]: EP2(bulk_out): device likely reset, stopping thread
Mar 18 00:44:29 host usb-proxy[1435]: End reading thread for EP02, thread id(1709)
Mar 18 00:44:29 host usb-proxy[1435]: event: disconnect
Mar 18 00:44:29 host usb-proxy[1435]: Resetting device
```

xairy, 2024-03-19:
usb-proxy has a quick to account for this dwc2 behavior, so I suspect your issue is something else.

Wrt the progress: I never got a response from the maintainers, so no progress.