xairy/raw-gadget #54: Facedancer Error w/ Raw Gadget

mcutshaw, 2023-11-08:
Currently attempting to run https://github.com/xairy/Facedancer/tree/rawgadget (rubber-ducky.py) on the latest raw-gadget dev branch and receiving the following error, thoughts?

    Traceback (most recent call last):
      File "/home/jack/Facedancer/rubber-ducky.py", line 37, in <module>
        main(device, type_letters())
      File "/home/jack/Facedancer/facedancer/devices/__init__.py", line 51, in default_main
        device.emulate(*coroutines)
      File "/home/jack/Facedancer/facedancer/future/device.py", line 172, in emulate
        self.connect()
      File "/home/jack/Facedancer/facedancer/future/device.py", line 124, in connect
        self.backend = FacedancerUSBApp()
                       ^^^^^^^^^^^^^^^^^^
      File "/home/jack/Facedancer/facedancer/core.py", line 23, in FacedancerUSBApp
        return FacedancerApp.autodetect(verbose, quirks)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/jack/Facedancer/facedancer/core.py", line 54, in autodetect
        return subclass(verbose=verbose, quirks=quirks)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/home/jack/Facedancer/facedancer/backends/rawgadget.py", line 345, in __init__
        super().__init__(device, verbose)
      File "/home/jack/Facedancer/facedancer/core.py", line 98, in __init__
        self.init_commands()
      File "/home/jack/Facedancer/facedancer/backends/rawgadget.py", line 352, in init_commands
        self.api.set_timeout(usb_raw_timeout_type.USB_RAW_TIMEOUT_EVENT_FETCH, 0, 20)
      File "/home/jack/Facedancer/facedancer/backends/rawgadget.py", line 281, in set_timeout
        RawGadgetRequests.USB_RAW_IOCTL_SET_TIMEOUT(self.fd, arg)
      File "/home/jack/Facedancer/facedancer/backends/rawgadget.py", line 172, in fn
        rv = fcntl.ioctl(fd, req, arg, True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    OSError: [Errno 22] Invalid argument
    sys:1: RuntimeWarning: coroutine 'type_letters' was never awaited

xairy, 2023-11-09:
Please double-check that you're using the Raw Gadget module from the `dev` branch and not from `master`.

If it still doesn't work, please rebuild the Raw Gadget module with `define DEBUG` as [described in the Readme](https://github.com/xairy/raw-gadget#troubleshooting) and share the `dmesg` output.

Also, which UDC are you using?

xairy, 2023-11-16:
Just checked that everything works on RPi 4 for me. The only reason for getting the `Invalid argument` error that comes to mind is if you're using the `master` branch of Raw Gadget instead of `dev`. But if you don't, I'm not sure what's wrong. You can try adding debug printfs to `raw_gadget.c` to find out what exactly fails.

xairy, 2023-12-08:
Added a note in 0332ec70d0b31805675ef476e5ba8533deeeb45a.

closed 2023-12-08 by xairy