xairy/raw-gadget #56: Logitech webcam can not proxy with AristoChen/usb-proxy

imatespl, 2023-11-11:
hi,@xairy,
    I test the Logitech webcam, that can not proxy with AristoChen/usb-proxy, because the webcam have 4 interface and each interface has multi altsettings, and some altsettings has no endpoints, then I start  usb-proxy, when I click windows 11 webcam preview , the windows driver will send set altsetting control packet to webcam, and then usb-proxy will` terminate_eps` and restart  process thread `process_eps`, then the raw-gadget ioctl enable return` ioctl(USB_RAW_IOCTL_EP_ENABLE): Device or resource busy `
https://github.com/AristoChen/usb-proxy/blob/c681f1238371fae331482f7d300a4652586d0dd3/proxy.cpp#L474-L482
this line cause the issues
https://github.com/AristoChen/usb-proxy/blob/c681f1238371fae331482f7d300a4652586d0dd3/proxy.cpp#L267
the log is like this, first not start any thread for this zero endpoint altsetting, then terminate any thing and start thread for altsetting has endpoint when the windows driver send control message change altsetting
```
Activating 1 endpoints on interface 0
int_in: addr = 3, ep = #4
Creating thread for EP83
Start reading thread for EP83, thread id(1849)
process_eps done
Start writing thread for EP83, thread id(1850)
Activating 0 endpoints on interface 1
process_eps done
Activating 0 endpoints on interface 2
process_eps done
Activating 0 endpoints on interface 3
process_eps done

ep0: transferred 26 bytes (out)
event: control, length: 8
  bRequestType: 0x01 (OUT), bRequest: 0x0b, wValue: 0x000a, wIndex: 0x0001, wLength: 0
  type = USB_TYPE_STANDARD
  req = USB_REQ_SET_INTERFACE
Changing interface/altsetting
Activating 1 endpoints on interface 1
ioctl(USB_RAW_IOCTL_EP_ENABLE): Device or resource busy

```
 I also change the code bypass the zero endpoint altsetting, then first start thread will failed, the issues is same, ioctl(USB_RAW_IOCTL_EP_ENABLE): Device or resource busy
```
Creating thread for EP83
Start reading thread for EP83, thread id(4029)
process_eps done
Start writing thread for EP83, thread id(4030)
Activating 1 endpoints on interface 1
ioctl(USB_RAW_IOCTL_EP_ENABLE): Device or resource busy

```
The device is `c922 Pro Stream Webcam`
The windows 11 driver is   Logitech 2021/4/27 1.4.40.0
The usb-proxy run on orangepi pc use the driver is musb-hdrc and device is musb-hdrc.4.auto
See attachment for details
[lsusb-output.txt](https://github.com/xairy/raw-gadget/files/13325991/lsusb-output.txt)
[usb-proxy-console-log.txt](https://github.com/xairy/raw-gadget/files/13325992/usb-proxy-console-log.txt)

imatespl, 2023-11-13:
I think  this issue my same with https://github.com/AristoChen/usb-proxy/issues/13, I will try it

imatespl, 2023-11-17:
I try raw_gadget dev  branch and usb-proxy reset branch, there also has the issue

xairy, 2023-11-17:
Not sure what exactly fails in your case, but neither usb-proxy nor Raw Gadget have proper support for isochronous transfers, so proxying webcams or other AV devices will not work anyway.

imatespl, 2023-11-20:
Thank you very much. I am closing this issue.

closed 2023-11-20 by imatespl

xairy, 2024-06-15:
FTR, as it turns out, Raw Gadget does somewhat support isochronous transfers; see #23.