xairy/raw-gadget #57: Connect/disconnect in raw-gadget

fxsheep, 2023-11-23:
Is there any way to initiate a connect/disconnect in raw-gadget?

xairy, 2023-11-23:
Could you elaborate what kind of connect/disconnect you mean? The `USB_RAW_IOCTL_RUN` ioctl is the one that starts the enumeration process, but I'm not sure if that's what you're asking.

fxsheep, 2023-11-24:
Sorry for the ambiguity.
Currently my understanding is, if I want the virtual device to 'disappear', I'll have to close the `fd`. To enumerate again, I need to start over by opening the device node, then invoke `USB_RAW_IOCTL_INIT` and `USB_RAW_IOCTL_RUN`.
I mean if it's possible to simulate a disconnect (analogue to disabling pull-up in real hardware), without closing the `fd`?

xairy, 2023-11-24:
Ah, I see. No, this is not currently possible. It might be possible to implement this, but I'm wondering: what's the use case for this?

fxsheep, 2023-11-24:
There isn't a specific use case, I just wondered if that's the optimal way(for now) to initiate a disconnect. Thanks for clarifying.

closed 2023-11-24 by fxsheep