xairy/raw-gadget #72: Support isochronous transfers in Dummy HCD/UDC

xairy, 2024-06-16:
This is not an issue in Raw Gadget, but it is related to its testing and usage.

The Dummy HCD/UDC does not support isochronous transfers. As a result, it's impossible to use it for connecting devices that require them, such as audio/video devices.

There was a [patch](https://lore.kernel.org/linux-usb/20220126132249.2133168-1-m.grzeschik@pengutronix.de/) for adding partial support for this, but it was rejected, as the support did not emulate the real hardware behavior; see the discussion on the patch for details.

xairy, 2024-06-19:
Alan posted a testing-grade [patch](https://lore.kernel.org/linux-usb/82b983a3-7ed5-4800-8a3d-1615c381d115@rowland.harvard.edu/T/#u) from pre-2008 era for this.

GPlays09, 2024-07-09:
since this isn't supported yet, is there a way to check if there is any data waiting to be read at a specific endpoint?

xairy, 2024-07-09:
I think you can do this via `usb_ep_fifo_status`, but it's not exposed in Raw Gadget (#12). Also the Dummy UDC does not appear to implement the `fifo_status` handler, so `usb_ep_fifo_status` won't work with it as is: you would need to add support for `fifo_status` to the Dummy UDC first. But even if you do, you still cannot use it for isochronous transfers before those are implemented.

But in case your question is not related to isochronous transfers and your goal is just to deal with Raw Gadget blocking on IO, what you can do is create a thread for handling I/O on an endpoint and send any signal to that thread (via e.g. `pthread_kill`) to interrupt the I/O ioctl when needed.