xairy/raw-gadget #74: keyboard example - fail: no response

GPlays09, 2024-07-06:
After building the example on a Pi4 and running it like so: sudo ./keyboard fe980000.usb fe980000.usb
Pi4 is connected to a windows PC.

I get this output: 
`event: connect, length: 0
ep #0:
  name: ep1in
  addr: 1
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #1:
  name: ep1out
  addr: 1
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #2:
  name: ep2in
  addr: 2
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #3:
  name: ep2out
  addr: 2
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #4:
  name: ep3in
  addr: 3
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #5:
  name: ep3out
  addr: 3
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #6:
  name: ep4in
  addr: 4
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #7:
  name: ep4out
  addr: 4
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #8:
  name: ep5in
  addr: 5
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #9:
  name: ep5out
  addr: 5
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #10:
  name: ep6in
  addr: 6
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #11:
  name: ep6out
  addr: 6
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #12:
  name: ep7in
  addr: 7
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #13:
  name: ep7out
  addr: 7
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep_int_in: addr = 1
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 64
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x200, wIndex: 0x0, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
config->wTotalLength: 34
ep0: transferred 34 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x302, wIndex: 0x409, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x300, wIndex: 0x0, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x301, wIndex: 0x409, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 18
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE
ep0: transferred 18 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x200, wIndex: 0x0, wLength: 9
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
config->wTotalLength: 34
ep0: transferred 9 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x200, wIndex: 0x0, wLength: 34
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_CONFIG
config->wTotalLength: 34
ep0: transferred 34 bytes (in)
event: control, length: 8
  bRequestType: 0x0 (OUT), bRequest: 0x9, wValue: 0x1, wIndex: 0x0, wLength: 0
  type = USB_TYPE_STANDARD
  req = USB_REQ_SET_CONFIGURATION
ep0: ep_int_in enabled: 0
ep0: spawned ep_int_in thread
ep0: transferred 0 bytes (out)
event: control, length: 8
  bRequestType: 0x21 (OUT), bRequest: 0xa, wValue: 0x0, wIndex: 0x0, wLength: 0
  type = USB_TYPE_CLASS
  req = HID_REQ_SET_IDLE
ep0: transferred 0 bytes (out)
event: control, length: 8
  bRequestType: 0x81 (IN), bRequest: 0x6, wValue: 0x2200, wIndex: 0x0, wLength: 129
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  descriptor = HID_DT_REPORT
ep0: transferred 65 bytes (in)
ep_int_in: key down: 8
ep_int_in: key up: 8
event: control, length: 8
  bRequestType: 0x21 (OUT), bRequest: 0x9, wValue: 0x200, wIndex: 0x0, wLength: 1
  type = USB_TYPE_CLASS
  req = HID_REQ_SET_REPORT
ep0: transferred 1 bytes (out)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x301, wIndex: 0x409, wLength: 1026
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x302, wIndex: 0x409, wLength: 1026
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x300, wIndex: 0x0, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x301, wIndex: 0x409, wLength: 255
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_STRING
ep0: transferred 4 bytes (in)
ep_int_in: key down: 8
ep_int_in: key up: 8
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x600, wIndex: 0x0, wLength: 10
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE_QUALIFIER
ep0: transferred 10 bytes (in)
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x700, wIndex: 0x0, wLength: 9
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_OTHER_SPEED_CONFIG
fail: no response`

One 'X' is typed and this errors shows up

xairy, 2024-07-06:
Hi! Is this Windows 11? Windows 10 never asks

Could you try this patch:

``` c
diff --git a/examples/keyboard.c b/examples/keyboard.c
index 0ac8680..c593f51 100644
--- a/examples/keyboard.c
+++ b/examples/keyboard.c
@@ -575,7 +575,7 @@ struct hid_descriptor usb_hid = {
        },
 };
 
-int build_config(char *data, int length) {
+int build_config(char *data, int length, bool other_speed) {
        struct usb_config_descriptor *config =
                (struct usb_config_descriptor *)data;
        int total_length = 0;
@@ -607,6 +607,9 @@ int build_config(char *data, int length) {
        config->wTotalLength = __cpu_to_le16(total_length);
        printf("config->wTotalLength: %d\n", total_length);
 
+       if (other_speed)
+               config->bDescriptorType = USB_DT_OTHER_SPEED_CONFIG;
+
        return total_length;
 }
 
@@ -756,7 +759,12 @@ bool ep0_request(int fd, struct usb_raw_control_event *event,
                        case USB_DT_CONFIG:
                                io->inner.length =
                                        build_config(&io->data[0],
-                                               sizeof(io->data));
+                                               sizeof(io->data), false);
+                               return true;
+                       case USB_DT_OTHER_SPEED_CONFIG:
+                               io->inner.length =
+                                       build_config(&io->data[0],
+                                               sizeof(io->data), true);
                                return true;
                        case USB_DT_STRING:
                                io->data[0] = 4;
```

GPlays09, 2024-07-06:
Ok, my bad after restarting both Pi and PC the issue seemed to resolve, PC is running Windows 10 and after restart the original Keyboard.c worked, also I decided to test the suggested changes which worked as well.

closed 2024-07-06 by GPlays09