xairy/raw-gadget #75: adding new  SBC compatibility

daytona675, 2024-07-12:
Hi xairy, 
thanks for you tool and your work.
I would like to inform you that I've tested raw_gadget on my new RPI5 8Go.
It work fine ! 
So you can add it in your readme on the section board compatibility.
Thanks again for your work ! 
K.R

daytona675, 2024-07-12:
for this section:USB Device Controllers ;)

xairy, 2024-07-17:
Hi! That's awesome, thank you for testing! Could you tell me the kernel version that you used and the names of the UDC device and driver?

daytona675, 2024-07-17:
Hi xairy,
for kernel version i've tested:
6.8.0-1007-raspi 
6.5.x branch
6.2x branch but I'm not sure for this one

For UDC device and driver:
I'm using this driver for both device and driver:
cat /sys/class/udc/1000480000.usb/uevent 
USB_UDC_NAME=1000480000.usb
K.R

commit 52f5ca72 (2024-07-17): readme: add Raspberry Pi 5 to UDC table

Fixes #75.
commit 71c796b9 (2024-07-17): readme: add Raspberry Pi 5 to UDC table

Fixes #75.
closed 2024-07-17 by xairy