xairy/raw-gadget #78: Possibility of emulating a USB hub for multiple devices?

Berghopper, 2024-08-27:
Not super familiar with the USB spec. But I do know that USB hubs have their own protocol.
Would it be anyhow possible to emulate different devices with raw_gadget?
(not just adding more endpoints, but actually different devices)..

In some cases this might be very useful and avoid having to use multiple UDCs.

xairy, 2024-08-27:
Emulating a USB hub is only possible with an [Aspeed UDC](https://www.spinics.net/lists/linux-usb/msg157518.html). I never tried this myself though. Other UDCs just don't support this.

But emulating any type of a normal device should be possible minding [the existing limitations](https://github.com/xairy/raw-gadget?tab=readme-ov-file#limitations) without any kernel changes.

closed 2024-08-27 by xairy

Berghopper, 2024-08-29:
Thanks for answering! Working on a project to emulate multiple 360 controllers, hence why the question :).

My next and last resort would be the 360 wireless receiver, since it actually acts as 1 device, but can support up to 4 controllers.

xairy, 2024-08-29:
You can technically emulate a single USB device with up to 255 HID interfaces, each of which acting as a separate input source. That's what that 360 wireless receiver likely does. Maybe this will work for your use case.

Berghopper, 2024-08-29:
@xairy Seems to work indeed:

https://github.com/Berghopper/360-w-raw-gadget/

It's not quite recognized as two separate devices, but xinput definitely registers 2 control surfaces/controllers! :)

Also I had to change the wireless receiver's protocol to that of a 'normal' controller (little bit of cheating there).

xairy, 2024-08-30:
That's awesome! It's great to see Raw Gadget being useful!

Thanks for noting that FunctionFS does not support providing that vendor-specific descriptor. I wasn't aware of this limitation.