xairy/raw-gadget #80: Raspberry Pi Zero 2W fail to run examples

acmileZZZ, 2024-11-08:
I followed instructions to set up for my Raspberry Pi Zero 2W(2024-03-12-raspios-bookworm-arm64)
When i connect the OTG port to my windows10 and run Keyboard, but windows show “Unknown USB Device（Device Descriptor”
The following is the output, it stucked at the last line.
```mimi@mimi:~/raw-gadget-master/examples $ sudo ./keyboard  fe980000.usb fe980000.usb
event: connect, length: 0
ep #0:
  name: ep1in
  addr: 1
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #1:
  name: ep1out
  addr: 1
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #2:
  name: ep2in
  addr: 2
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #3:
  name: ep2out
  addr: 2
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #4:
  name: ep3in
  addr: 3
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #5:
  name: ep3out
  addr: 3
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #6:
  name: ep4in
  addr: 4
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #7:
  name: ep4out
  addr: 4
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #8:
  name: ep5in
  addr: 5
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #9:
  name: ep5out
  addr: 5
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #10:
  name: ep6in
  addr: 6
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #11:
  name: ep6out
  addr: 6
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep #12:
  name: ep7in
  addr: 7
  type: iso blk int
  dir : in  ___
  maxpacket_limit: 1024
  max_streams: 0
ep #13:
  name: ep7out
  addr: 7
  type: iso blk int
  dir : ___ out
  maxpacket_limit: 1024
  max_streams: 0
ep_int_in: addr = 1
event: control, length: 8
  bRequestType: 0x80 (IN), bRequest: 0x6, wValue: 0x100, wIndex: 0x0, wLength: 64
  type = USB_TYPE_STANDARD
  req = USB_REQ_GET_DESCRIPTOR
  desc = USB_DT_DEVICE```

xairy, 2024-11-13:
Could you try it with a Linux host and see if that works?

acmileZZZ, 2024-11-16:
No, Ubuntu20.04 nothing happened.
"lsusb" nothing shows
"dmesg" no logs for new pluged one

acmileZZZ, 2024-12-18:
Solved. Removed my extend board, Example scripts works well.

closed 2024-12-18 by acmileZZZ