xairy/raw-gadget #84: Resolve 4K limit on control transfers

msbroadf, 2025-07-17:
There seems to be an easy resolution to the 4k limitation on control data

You can change copy_from_user to vmemdup_user and kmalloc to kvmalloc (and associated kvfrees) and that seems to work ok for larger transfers

xairy, 2025-07-22:
We should be able to simply increase the limit on the transfer size and the current code should work as is (assuming the UDC sybsystem can handle this, but I think it should).

The question is whether this is actually required? Are there cases when we want to send more than 4K bytes in a single transfer?

xairy, 2025-09-04:
Patch [sent](https://lore.kernel.org/linux-usb/a6024e8eab679043e9b8a5defdb41c4bda62f02b.1757016152.git.andreyknvl@gmail.com/T/#u).

xairy, 2025-10-27:
Patch [merged](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=37b9dd0d114a0e38c502695e30f55a74fb0c37d0).